sensebox/books-v2 #46: Translate :home & :edu Book into English

synolus-david, 2018-07-30:
It's a torture but necessary ;-) 

A task for the summer?

@buchwaldchassee 
@umut0 
@jasperbuss 
@Thiemann96 
@bkari02

Thiemann96, 2018-08-02:
Uff... hat da nicht jemand @mariopesch ? schon was zu gemacht ?

Thiemann96, 2018-08-09:
Wollten abwarten bis das deutsche Book komplett fertiggestellt ist und dann in einem Sprint das ins Englische übersetzen. Die "Architektur" würden wir anlehnen an das [Blockly Buch](https://sensebox.github.io/books-v2/blockly/).

bkari02, 2018-09-17:
Home Buch wurde übersetzt. 
Bitte alle einmal durchlesen und Fehler beheben, bzw Issues und Pull-Request stellen. 
[Hier der Link zum Buch](https://sensebox.github.io/books-v2/home/)

@buchwaldchassee 
@umut0 
@jasperbuss 
@Thiemann96 
@synolus-david 
@mariopesch 
@bgunt

closed 2019-03-07 by bkari02